YITechnology/YIOpenAPI #1: Possible bug in Yi 4k firmware

Code43210, 2016-09-02:
Hi,

My firmware version: Z16V13L_1.1.0_en-us_build-20160802190839_git-5222f2b7_r2139

When requesting the possible values for "photo_stamp" then the following is returned from the camera:

`{"msg_id":9,"options":["off","time","date","date/time"],"param":"precise_cont_time","permission":"settable","rval":0}`

I would expect this:

`{"msg_id":9,"options":["off","time","date","date/time"],"param":"photo_stamp","permission":"settable","rval":0}`

All other settings seem to work fine.

Code43210, 2016-09-02:
The camera locate command does not work neither:
`{"msg_id":2,"param":"on","token":2,"type":"buzzer_ring"}`

taoma618, 2016-09-13:
Thanks, we are looking into it