YITechnology/YIOpenAPI #2: Viewfinder problem when switching between modes

Code43210, 2016-09-19:
Hi,

This problem happens only on the Xiaomi Yi 4k (not on the old normal model):

1.) Viewfinder is off.
2.) Now switch between modes (video, photo, timelapse, etc.).
3.) Now try to turn on the viewfinder. Sometimes it works, but often the viewfinder does not start but instead an error -21 is returned.

After this procedure also the original Xiaomi Yi app for Android shows only a black viewfinder until the camera is manually rebooted.

Regards,

xieyubo, 2016-09-21:
-21 means camera is busy. Please try it later.

Code43210, 2016-09-21:
Ok thanks. But it seems as if the camera hangs, e.g. it does not allow to enable the viewfinder anymore. Changing of other settings work thus.

xieyubo, 2016-09-22:
Yes, some settings can only be set when camera is idle. Some settings don't have this limitation.

Code43210, 2016-09-22:
This means that the only solution is to restart the camera when it gets into this "non-idle" state?

xieyubo, 2016-09-22:
No, wait for 1  or 2 seconds after taking action, camera can be in idle state. In most cases, busy means the previous action is not finished on camera side.

closed 2016-09-29 by xieyubo