YITechnology/YIOpenAPI #4: Disable all Log.info

bknill, 2016-09-20:
Any chance you could release code with all Log.info removed, I'm trying to implement in a Cordova App and it's very hard to overide the initialisation function.

xieyubo, 2016-09-21:
Hi, Platform.initialize() is required. You must invoke it before using any other YICamera SDK functions. But this function is not required to invoke in onCreate(). You can invoke it in other place. You only need make sure it will be invoked at least once before invoking other YICamera SDK functions.

closed 2016-09-21 by xieyubo

bknill, 2016-09-21:
Hi, The problem is that I'm using this in a Cordova Plugin and I don't seem to be able to access the Platform.initialize() override, I think it's being initialised elsewhere or before this code is being called. As far as I can see, you only use it for 

```
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        // Initialize YiCameraPlatform
        try {
            Platform.initialize(new Logger() {
                @Override
                public void verbose(String message) {
                    Log.v("YiCameraPlatform", message);
                }

                @Override
                public void info(String message) {
                    Log.i("YiCameraPlatform", message);
                }

                @Override
                public void warning(String message) {
                    Log.w("YiCameraPlatform", message);
                }

                @Override
                public void error(String message) {
                    Log.e("YiCameraPlatform", message);
                }
            });
        } catch (Exception ex) {
        }

        StrictMode.ThreadPolicy policy = new StrictMode.ThreadPolicy.Builder().permitAll().build();
        StrictMode.setThreadPolicy(policy);

        pendingCameras = new ArrayList<>();
        cameras = new ArrayList<>();
        gridAdapter = new GridAdapter(this, cameras);

        setContentView(R.layout.grid_activity);
        gridView = (GridView) findViewById(R.id.gridView);
        gridView.setAdapter(gridAdapter);
    }
```

Which is setting the initial variables and adding those useless extra Log functions. I just need to remove those useless Log functions, i.e. make them all Log.i, or Log,d etc so it can use the generic Android Logger.

This seems like such a silly thing to block people from using this API from, it's not doing anything, just Logging messages - there is no obvious good reason why Log.info can't just stay as Log.i in your code!

bknill, 2016-09-21:
Alternatively can you give access to the source code so I can just remove them?

xieyubo, 2016-09-22:
The above code is just sample code in our sdk, you can modify it.

bknill, 2016-09-22:
It's the source of the Jar I need to modify, just to remove the Log.infos, can I have access to that?

xieyubo, 2016-09-22:
Sorry, I know little about cordova. But the file is here: https://github.com/YITechnology/YIOpenAPI/blob/master/sdk/java/samples/YI360Demo/app/src/main/java/com/xiaoyi/yi360demo/GridActivity.java. I think you can modify it, right?

bknill, 2016-09-25:
Yes.. That is what I have been doing. But please read the following very carefully. 

1.  You have added these overrides in the Platform.initialisation method. I cannot override this in Cordova.  The plugin gets loaded after the platform is initialised. 
2. The Log override is redundant, it's changing Log.i to Log.info. It's a very small change that does very little. 
3. This changing of the Log completely stops me using the API. I can connect to the cameras, but anytime anything happens with a Log.info it crashes. 

I have completely given up using your API because of this. It's incredibly frustrating that something so small as a Log.info as opposed to the usual Log.i is completely blocking me. 

If I could have the source code, I could change all the Log.infos to Log.i's and it would be fine.  At the moment, the API is unusable.

xieyubo, 2016-09-27:
Can you paste your code? I don't understand what you want to do. Platform.initializ() can ben invoked at any time. And this Platform is our sdk platform, not Cordova platform or Android SDK platform.

bknill, 2016-09-27:
Ahh OK. That's where I was going wrong.  I couldn't find where Platform.initialise() was coming from, Thanks. I'm in business.