YITechnology/YIOpenAPI #7: can not connect my wifi router

deepzf, 2016-09-24:
when i used firmware and configed the file "sta.conf", i could not connect my wifi route ,i checked my configration again and again,but i could not find any error .I used notepad for windows to config it ,was that right?

taoma618, 2016-09-26:
you did it correctly. but there is a known issue that the editing under windows environment would introduce an extra 0x0d for the linefeed which might end of causing problem in our firmware when reading the file. 

we are fixing our firmware now to make it more robust on such situations. in the meanwhile, please try 1) using a binary editor to remove the 0x0d from your sta.conf file; or 2) editing it under a non-windows environment.

let us know how that goes.

deepzf, 2016-09-27:
ok,i'll try it

closed 2016-09-29 by xieyubo