YITechnology/YIOpenAPI #8: Yi disconnects after starts recording

bknill, 2016-09-27:
OK now I've got the app running I'm trying to start recording

The camera is connected fine, it's got the latest firmware (1.2.13)

Here is the log

`V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onDataSent(),
 current state: Connected
I/XiaomiYiCordovaPlugin(16988): Start recording success
V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onDataReceive
d(), current state: Connected
V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onDataSent(),
 current state: Connected
V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onDataReceive
d(), current state: Connected
V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onDataSent(),
 current state: Connected
V/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] internalDisco
nnect(), current state: Connected
I/XiaomiYiCordovaPlugin(16988): [com.xiaoyi.action.ActionCamera@423fe5c0] onClosed(), c
urrent state: Disconnecting
I/XiaomiYiCordovaPlugin(16988): Start recording failed
I/XiaomiYiCordovaPlugin(16988): camera connection closed`

Here is my recording function (copied from the example)
`  public void startRecording(Date startTime,final CallbackContext callbackContext) {
    if (mState == CameraState.Connected) {
      updateState(CameraState.StartRecording);
      mCamera.stopRecording(null, null).setSystemMode(SystemMode.Record, null, null)
        .startCommandGroup()
        .setDateTime(new Date(), null, null)
        .startRecording(startTime.getHours(), startTime.getMinutes(), startTime.getSeconds(), null, null)
        .submitCommandGroup(null, new ActionCameraCommandCallback1<Error>() {
          @Override
          public void onInvoke(Error val) {
            Log.i(TAG, "Start recording failed");
            if (mState == CameraState.StartRecording) {
              updateState(CameraState.Connected);
            }
          }
        });
      Log.i(TAG, "Start recording success");
      PluginResult result = new PluginResult(PluginResult.Status.OK, true);
      callbackContext.sendPluginResult(result);
    }
  }`

Here is my camera setup

`  public Camera( String ip, CallbackContext callbackContext) {
    mUIHandler = new Handler(Looper.getMainLooper());
    mState = CameraState.Disconnected;
    mCallbackContext = callbackContext;
    mIP = ip;
    if (mIP == null) {
      mIP = "";
    }
/*    mHostname = hostname;
    if (mHostname == null || mHostname.isEmpty()) {
      mHostname = mIP;
    }*/
  }`

Is this something to do with hostname? I could see where that was being used.

bknill, 2016-09-27:
it appears to be after you attempt to change mode

I/XiaomiYiCordovaPlugin(20195): setMode video
V/XiaomiYiCordovaPlugin(20195): [com.xiaoyi.action.ActionCamera@4233f4c8] onDataSent(), current st
ate: Connected
V/XiaomiYiCordovaPlugin(20195): [com.xiaoyi.action.ActionCamera@4233f4c8] internalDisconnect(), cu
rrent state: Connected

bknill, 2016-09-28:
I've tried this approach
`      mCamera.setDateTime(new Date(), null
        ,new ActionCameraCommandCallback1<Error>()
        {
          @Override
          public void onInvoke (Error val){
          Log.i(TAG, "setting date failed" + val.getCode());
        }}).startRecording(startTime.getHours(), startTime.getMinutes(), startTime.getSeconds(), null, new ActionCameraCommandCallback1<Error>() {
        @Override
        public void onInvoke(Error val) {
          Log.i(TAG, "start recording failed"  + val.getCode());
        }
      });`

And it results in this error code but there is no error detail

`I/XiaomiYiCordovaPlugin( 8605): setting date failed-2147483641
V/XiaomiYiCordovaPlugin( 8605): [com.xiaoyi.action.ActionCamera@4236e638] onDataSent(), currentstate: Connected
V/XiaomiYiCordovaPlugin( 8605): [com.xiaoyi.action.ActionCamera@42939110] onDataReceived(),current state: Connected
V/XiaomiYiCordovaPlugin( 8605): [com.xiaoyi.action.ActionCamera@42939110] onDataSent(), currentstate: Connected
V/XiaomiYiCordovaPlugin( 8605): [com.xiaoyi.action.ActionCamera@4236e638] onDataReceived(),current state: Connected
I/XiaomiYiCordovaPlugin( 8605): start recording failed-2147483641
V/XiaomiYiCordovaPlugin( 8605): [com.xiaoyi.action.ActionCamera@42939110] onDataReceived(),current state: Connected
I/XiaomiYiCordovaPlugin( 8605): start recording failed-2147483641`

bknill, 2016-09-28:
Ahh.. does this SDK support the old Yi?

xieyubo, 2016-09-28:
What do you mean old YI? The sdk only support YI 4K.

bknill, 2016-09-28:
OK the opening statement on this repository

"YI Open API provides a set of APIs in the form of SDKs of different languages for software developers and hardware makers to build cool apps and products with **YI Action Camera** or YI 4K Action Camera."

Makes that confusing. I'm trying to get this to work with the YI Action Camera.  It kind of works but has a load of problems. If this isn't the case, and you only support the Yi4K. You need to say that.

taoma618, 2016-09-28:
Fixed the statement, good catch!

bknill, 2016-09-29:
OK so you have no plans for this SDK to support the Yi? Even though it seems quite close to be able to.. i.e. it will connect and seems to try and start recording. 

I've reverted to the reverse engineered code other people have used for the Yi. I really wanted to use the functions you have in here to set the time to start recording, for better synchronization.  

It's a shame because this Cordova Plugin is mostly working and I was planning on releasing in next week once I had recording and photos working.

xieyubo, 2016-09-29:
The firmware of YI (not YI 4K) doesn't support set the time to start recording. We don't maintain that firmware now. So that feature won't be supported on YI.

closed 2016-09-29 by xieyubo