YITechnology/YIOpenAPI #10: Can't build on iOS (Yi360Demo)

marcos10soares, 2016-09-28:
When trying to build Yi360Demo for iOS it gives the following errors:

'ActionCameraListener' is unavailable: cannot find Swift declaration for this class
'Error' is unavailable: cannot find Swift declaration for this class
'Super' is unavailable: cannot find Swift declaration for this class

Method does not override any method from its superclass Camera.swift (regarding override func onConnected)

Tried with Xcode:
8.1
8.0
7.3.1

iOS - 9.3.4

Thanks

marcos10soares, 2016-09-28:
I deleted the frameworks (CocoaASyncSocket, YiCamerasSDK, SwiftyJSON) and dragged them again to the project and now was able to build.

But when running I only get a blank app screen and a fatal error, the console prints:
dyld: Library not loaded: @rpath/CocoaAsyncSocket.framework/CocoaAsyncSocket
  Referenced from: /var/containers/Bundle/Application/56BA9053-9FAD-4987-A290-1430C365AAFE/YI360Demo.app/YI360Demo
  Reason: image not found
(lldb)

xieyubo, 2016-09-29:
You need add CocoAsyncSocket.framework to your library search path.

marcos10soares, 2016-09-29:
@xieyubo thank you, but I added "$(SRCROOT)/YI360Demo" to the library search path and still have the blank screen and "lldb" on the console.

dyld`dyld_fatal_error:
->  0x120059088 <+0>: brk    #0x3

JaviLorbada, 2017-02-03:
@marcos10soares I got it working with the following search path in the build settings of the project

[image]

I hope with that works :)

PR #31 (2017-02-19): Fixes YI360Demo project.
## What's in this pull request?

### Description
I've tried to fix several issues while running YI360Demo project. I describe below the issues one and one.

- Remove all warnings, especially the ones about unused result and remove all `;` to make more swift-ish. 9348f5f
- Fix issue on CameraSelectionViewController not sending the `cameraCount` to the CameraGridViewController a753d09
- Fix crash trying to access index out of bounds. 7d5a227
- Fix range of IPs since connection 4 cameras or more, most of them were starting on 172.20.10.11-12-13 and so on... 	d4c2ea2
- Fix [... cut]
commit ee83a8fd (2017-02-19): Includes `$(SRCROOT)/CocoaAsyncSocket.framework` on Library Search Paths Build Settings, Fixes #10
closed 2017-02-19 by xieyubo