YITechnology/YIOpenAPI #17: (probably) Downloading issue

NikolayRag, 2016-10-25:
While parsing Java libyiaction.jar i've noticed
>           String url = "http://" + ActionCamera.this.mCameraIP + "/DCIM/100MEDIA/" + fileName;
string in ActionCamera.downloadFile().

But if you ever shot more than 999 times, new files will be placed in /101MEDIA/, then /102MEDIA/ for 1999+ files and so on. And in that case files could not be readed AT ALL by the function.

_Humble but vital petition from here:
Could you please please add ability to access file by HTTP from particular offset, something like "http://...mp4?offset=xxxbytes"? I believe it costs nothing for you..._

rahmatnazali, 2017-06-06:
@NikolayRag Hi, I'm having the same issue :)
Not tested yet, but I'm thinking of making a new class that extends **ActionCamera.class**, and overide the `public ActionCamera downloadFile` function. Then maybe we can make a proper `String url` on our own.
I hope this helps.

Water-bamboo, 2017-06-15:
Please wait, let me check for this. @NikolayRag

knkpxt, 2017-06-19:
Also cought the same bug. It would be great to have access to entire DCIM directory

NikolayRag, 2017-06-28:
This should be mentioned somewhere: Yi DID made partial HTTP download with 'Range' header.
Very thanks Yi.

rahmatnazali, 2017-07-12:
Hi!

I also got the same bug when the captured image starts to placed outside of `/100MEDIA/` directory.

Say, I have `/100MEDIA/` that is actually empty, and I have several captured image placed on `/101MEDIA/`.

The function `ActionCamera.getFileList` will return `NULL`, possibly because the API only search in `/100MEDIA/` as I find these codes inside the API : `return super.getData().put("param", "/tmp/fuse_d/DCIM/100MEDIA");`

knkpxt, 2017-08-08:
NikolayRag, have you found solution to this problem?

> This should be mentioned somewhere: Yi DID made partial HTTP download with 'Range' header.
Very thanks Yi.

Can you explain how this information can help me to get access to different directories?

NikolayRag, 2017-08-15:
Surely. I always download files by HTTP, not by API :)
Theres no problem with this, as file IS read by HTTP inside API itself, then sent to client.
And speaking of deletion and listing files, i made Python API (very quite different), that accept full path.

Happiness of partial download is related to my original 
> Humble but vital petition
It's unrelated to path issue.

closed 2021-08-07 by NikolayRag