YITechnology/YIOpenAPI #21: can't get rtsp to work

lrhymes, 2016-11-07:
I can get the rtsp url (it claims it is rtsp://192.168.42.1/live) and the camera ip is 192.168.1.199, but I can not view the stream in my app.  I tried just starting the rtsp server and using VLC and it also will not work.

Any ideas on what I'm doing wrong?

lrhymes, 2016-11-07:
Looks like rtsp://192.168.1.199/live has the stream.  Why does getrtspurl() return the wrong url?  '

I can log onto the hotspot with my laptop and view it with VLC.  Unfortunately my phone locks up when trying to view it.

xieyubo, 2016-11-08:
The url returned by getrtspurl() comes from firmware. There may be a bug in the firmware. We will check it.

lrhymes, 2016-11-13:
works using vitamio library

closed 2016-11-13 by lrhymes