YITechnology/YIOpenAPI #22: could not init platform

deepzf, 2016-11-09:
when i execute my program,it post some error :

Exception in thread "main" java.lang.NoClassDefFoundError: com/xiaoyi/action/Platform
	at Yi4k.main(Yi4k.java:224)
Caused by: java.lang.ClassNotFoundException: com.xiaoyi.action.Platform
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)

my codes is here,is there any error:

import com.xiaoyi.action.*;
import com.xiaoyi.action.Error;
import java.io.*;
import java.net.*;
import java.util.*;
import java.util.Date;

enum CameraState {
    Disconnected,
    Connected,
    StartRecording,
    Recording
}

class Camera extends ActionCameraListener {

    private CameraState mState;
    private ActionCamera mCamera;
    public String mIP;
    public String mHostname;

    //region Properties
    public CameraState getState() {
        return mState;
    }

    public String getIp() {
        return mIP;
    }

    public String getHostname() {
        return mHostname;
    }
    //endregion

    public Camera() {
        mState = CameraState.Disconnected;
    }

    public void connect() {
        mCamera = new ActionCamera(this, new DispatchQueue() {
            @Override
            public void dispatch(Runnable task) {
            }
        });
        mCamera.connect("tcp:" + mIP + ":7878");
    }

    public void disconnect() {
        if (mCamera != null) {
            mCamera.disconnect();
        }
    }

    public void startRecording(Date startTime) {
    }
    private void capturesuccess(){
        
    }
    public void startcapture(){
        if (mState == CameraState.Connected){
            updateState(CameraState.StartRecording);
            mCamera.stopViewFinder(null, null).stopRecording(null, null).setSystemMode(SystemMode.Record, null, null);
            mCamera.capturePhoto(null, null);
        }
    }
    public void stopRecording() {
        if (mState == CameraState.StartRecording || mState == CameraState.Recording) {
            mCamera.stopRecording(null, null);
        }
    }

    //region implementation of ActionCameraListener
    @Override
    public void onConnected() {
        updateState(CameraState.Connected);
        }
    }
    @Override
    public void onClosed(Error error) {
        updateState(CameraState.Disconnected);
    }

    @Override
    public void onRecordStarted() {
        updateState(CameraState.Recording);
    }

    @Override
    public void onRecordStopped() {
        updateState(CameraState.Connected);
    }
    //endregion

    private void updateState(CameraState state) {
        if (mState != state) {
            mState = state;
            //    mContext.gridAdapter.notifyDataSetChanged();
        }
    }
    public void doscan() {
    }
}

public class Yi4k {

    /**
     * @param args the command line arguments
     */
    private String s;
    

    public static void main(String[] args) {
        // TODO Auto-generated method stub
        
        try {
            Platform.initialize(null);
        } catch (Exception ex) {
            ex.printStackTrace();
        }
        Camera camera=new Camera();
        System.out.println("doscan......");
        camera.doscan();
        //camera.mIP = s;
        System.out.println("connecting......");
        camera.connect();
        
    } 
}

cftong, 2016-11-09:
I think the framework does not support native Java. It only works on the Android

deepzf, 2016-11-09:
This is a pity, I think a more generic sdk will win more Industry customers for you

closed 2016-11-09 by deepzf

xieyubo, 2016-11-09:
Supporting native java is in our plan. Will be ready soon.

deepzf, 2016-11-10:
I think RAST API will be better，widely used and easy maintenance