YITechnology/YIOpenAPI #23: Can't connect with camera

dureo, 2016-11-10:
I'm trying to conenct with a camera but onConnected() or onClosed() are never called.

I create ActionCamera like this
`ac1 = new ActionCamera(new ActionCameraListener() {
            @Override
            public void onConnected() {
                super.onConnected();
                connFeedback.setText("Connected to camera " + SSID1);
            }

            @Override
            public void onClosed(Error error) {
                super.onClosed(error);
                connFeedback.setText("Error connecting to camera " + SSID1 + ": " + error.toString());
            }
        }, new DispatchQueue() {
            @Override
            public void dispatch(Runnable runnable) {

            }
        });`

and try to connect with this camera using
`ac1.connect("tcp:" + ip + ":7878");`

i'm sure that ip is correct (I tested it) but nothing happens.

Anyone can help me? thanks

lrhymes, 2016-11-10:
You need to tell it where to send runnable.  The UI cause ur trying to update the UI in the callbacks? Noob here...am I right?

dureo, 2016-11-10:
Ok, now it works and I understand it better, thanks.

closed 2016-11-11 by xieyubo

Asad1234567, 2018-04-26:
Please tell me what u used inside disatch function?