YITechnology/YIOpenAPI #24: GCODE Files issue

bozpet, 2016-11-14:
Its possible to provide the STL files? I tried to use the Gcode files but my printer have a very different configuration... 

Best Regards

brucechase, 2017-04-10:
I have similar problem.  I cannot find someone that will print from the Gcode file.  When I used a program to convert to STL and sent the STL to a 3dprint shop, the shop found the STL to be corrupted and unprintable.

svenerbeck, 2017-04-10:
4 get it - asked same things on multiple ways. Mail to support - Facebook etc...
No one is giving a fuck about questions at this company...

daniel-hermann, 2017-06-07:
same problem. any updates?

brucechase, 2017-06-07:
I designed a 6-camera rig and had one 3D printed.  It works out pretty
well.  If you have it printed, you will have to glue each of the parts
together (I used a 5-minute epoxy) and buy some 5mm bolt for the hinges.
You may have to do a little filing due to tight tolerances on the fitting
of the parts.  The filing and gluing took me about a half hour to
accomplish.

Here is the contact (David Kriesberg) for the 3d printing.  He has the STL
file/design that works with his 3d printer.  David was VERY helpful to me
in getting this to work.

Dkrispis@hotmail.com

Here is the rig with a base ordered from Amazon
<https://www.amazon.com/gp/product/B01E8CWB6G/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1>.
I really like the base's stability.

[image: Inline image 1]

Bruce