YITechnology/YIOpenAPI #38: Creating a Multi Endpoint Cable. There's no pin 7 in micro usb.

nldroid, 2017-03-11:
The [wiring diagram](https://github.com/YITechnology/YIOpenAPI/blob/master/spec/Multi-endpoint%20Micro%20USB%20Cable.pdf) decribes that you have to connect all pins 2 together and all pins 7 together. I'm a bit confused because micro usb cables only have 5 pins. If i use the colors then i should connect all red: vcc (+5v) cables together and all black: ground cables together. Is that correct?

bknill, 2017-03-13:
You need that 7 pin cable, I've tried with a few other 7 pin usb cables and no luck.

nldroid, 2017-03-14:
Hi @bknill, Thanks for your reaction.

I've searched for 7-pin connectors and it looks that you're right. Each manufacturer has it's own 7-pin connector so it all depends on the place of the pins. I still hope i find one some day :)

akc2267, 2017-04-17:
@nldroid @bknill 
Have either of you succeeded in creating and testing a multi-endpoint cable? If so, does it support synchronized starts of 4k video, or just the officially supported modes ( 12MP photo, 2.5K video and 2.5K time lapse video)?

Sundrey, 2017-05-24:
Hello!
I use 7pin usb cable, but it not work. Test on 2 cams. 
Connect 2pin to 2pin and 7pin to 7pin, another contacts will be free
1 cam with first frameware, second with others frimware
I use this cable https://www.goandroid.co.in/wp-content/uploads/2014/05/oppo-find-7-7a-fast-charging-aa-2-of-7-710x399.jpg
What I can do wrong?

yokotoka, 2017-05-29:
Wtf?! Why YI using non-existent in the market 7-pin connector?! Why not standard 5-pin?

daniel-hermann, 2017-06-07:
is there any update on this? Anyone found out which 7-pin connector YI is using?

daniel-hermann, 2017-06-07:
In [a post on the yi community](http://forum.xiaoyi.com/forum.php?mod=viewthread&tid=37935) an admin posts a tutorial for a yi camera, where a detailed picture of the usb connector is posted. 
When I google-image-search the picture I find [this](https://www.amazon.co.uk/Water-Micro-Connector-Solder-Socket/dp/B00P0BOXYI)

guardfromua, 2017-07-25:
@Sundrey  where did u find it? I want to but it too)

daniel-hermann, 2017-07-25:
I bought them here: https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.0.0.4LSdzM&orderId=84515479253055&productId=2032446271

I recieved them a few days ago, I can check if they work, but unfortunately I might only find time for it in a week or two.

guardfromua, 2017-07-31:
@daniel-hermann any results on your work.
Do think this cable will work? 
https://world.taobao.com/item/555445224213.htm?fromSite=main&spm=a230r.1.14.46.ebb2eb2a3nwOd&ns=1&abbucket=4#detail

kevinra, 2017-08-23:
To those who tried 7 pin cable, have you checked the signals of each pin?

I am thinking to build my own sync cable from third party 7 pin cable (like the ones you all mentioned with URLS) since the official sync cable currently ships to US only. Once I have the cable with me, I am thinking to use oscilloscope to verify which pin is responsible for genlock signal.

I wonder if this works though. Thoughts?

patricksavill, 2017-11-26:
@kevinra did you get any further with determining the pin outs on the connector, and which provides the genlock signal?