YITechnology/YIOpenAPI #42: Cameras disconnect themselfs if 10 connected by internalDisconnect()

steosoft, 2017-04-27:
I'm writing Android application which connects to 10 Xiaomi Yi cameras. Each one of them has different IP assigned by custom script. They all connects to Wi-Fi and Android device is also in the same network.
Cameras connects without any problems but every 1-2 minutes some camera disconnects from device (not from network because for testing purposes I'm running ping command to cameras from my computer connected to the same network). If I had 5 cameras connected it was working without any problems.

`com.xiaoyi.action.Logger` show logs in that situation (one after one in the same second):
verbose: `[com.xiaoyi.action.ActionCamera@396e82ee] internalDisconnect(), current state: Connected`
info: `[com.xiaoyi.action.ActionCamera@396e82ee] onClosed(), current state: Disconnecting`

In `ActionCameraListener` in `public void onClosed(YICameraSDKError error)` error object has values:
`error.getDetail(): null`
`error.getCode(): -2147483638`

Water-bamboo, 2017-06-05:
What's your phone model? Can't confirm if it's the phone's wifi's issue when connect beyond 5 cameras.

steosoft, 2017-08-01:
Described problem was caused by poor quality chargers to which cameras was connected.
But if I use original Samsung chargers problem sometimes happens - not often, like every few hours but then all cameras are being disconnected by internalDisconnect().
Problem occurs on Lenovo TAB 2 A10-70L but I can test it on other device if I might help.