YITechnology/YIOpenAPI #44: Yi 4k+ Cameras Synchronisation Issue

LoomyBear, 2017-05-02:
Hi,

I'm trying to follow this tutorial on syncing Yi 4k+ cameras together.
http://open.yitechnology.com/vrcamera.html
And I'm having problems follow this tutorial to the its end.

Here's what I did:

1. Downloaded `firmware_first.bin` and `firmware_others.bin` from this repository https://github.com/YITechnology/YIOpenAPI/tree/master/vr/bin/en-us *
2. Copied the firmware files to SD cards.
3. Renamed both files to `firmware.bin`
4. Inserted the SD cards into the corresponding cameras
5. Turn the cameras on.
6. I was asked if I wanna _UPDATE THE FIRMWARE_ [ OK / CANCEL ]
7. Tapped OK
8. Get _FIRMWARE VERIFICATION FAILED_ message from both cameras

* PLEASE NOTE: the firmware links on the tutorial page are not working. You need to update them.

I suppose those particular firmwares were compiled strictly for 4k cameras and won't work on 4k+ models.

Would you please tell me how should I proceed with this issue?

Thank you!!

Steve132, 2017-06-19:
This issue also effects me

Water-bamboo, 2017-08-05:
Not support on 4K+, please don't try the 4K firmware.

Steve132, 2017-08-08:
What does this mean?

archenroot, 2017-08-08:
I didn't notice this issue already here, I also bought 2 4k+ cameras and was interested in GENLOCK sync. Unfortunately this capability is not available for 4k+ models, only 4k supports GENLOCK mechanism.

I provided a hint for doing this via NTP sync and synchronization in the software (custom written) based on frame timestamp. Question is if the timestamp is available for streaming and not only storing. I will test during this weekend....

On the other hand, if you are interested in syncing the camera (4k+) for recording, there are some hacks available if I am not wrong again with NTP sync and then executing "start recording" command from camera linux shell accessible via telnet protocol.

NOTE: I didn't tested yet all possibilities, but will report some results as I am ordered these cameras for stereo vision and now got stuck :-))), but there is always some way....

kyberkiller, 2017-09-19:
How is this going? Very important issue. Can you put some links, I'm just starting with this - certainly not buying anything before clear route to synced mixing.

noemis84, 2018-01-08:
I'm searching for the same solution: sync two YI 4k+ for a simple stereo recording. Has someone maybe some new informations?

periox, 2018-12-21:
It seems Yi team left the idea of genlock for their cameras, which is a shame as more of us needs to sync the cameras. The solution may be on the hardware level - and here is an example described on the cheap Yi Discovery camera synchronization. [http://3djournal.com/3d-photo-and-3d-video-cheap-with-yi-discovery-cameras-1-133](url) It is not the most elegant solution in the world (a software solution would be much better), but it seems it works and can help someone.