YITechnology/YIOpenAPI #49: Downloaded image files are corrupted

rahmatnazali, 2017-06-06:
Hi! :)

So I have a YI 4k Action Camera and using YI open API I'm trying to control the camera via android apps, capture a photo, and download the captured photo to external storage on my device.

All did well, until I try to download the image captured using `ActionCamera.downloadFile("YDXJXXXX.JPG")`. Here's the detail:
- The download finished successfully, as the callback shows that `DownloadTask.downloadBytes == DownloadTask.totalBytes`
- BUT, when I try to open the downloaded via file manager, it says that my image are corrupted
- After some observation, I find something strange. The original image captured (inside YI 4k Camera) was sized around 4MB, while downloaded image (inside external storage of my android device) was around 11MB!

Did I make a mistakes here? Please let me know.

Thanks in advance.

P.S. These are the codes to download the image:

```
public void downloadAFile(String fileName){
        String filePath = Constant.externalStoragePath;
        filePath = filePath + fileName;

        this.downloadFile(fileName, filePath, new ActionCameraCommandCallback1<DownloadTask>() {
            @Override
            public void onInvoke(DownloadTask downloadTask) {
                ambilFotoActivity.onCameraSucceedDownloadFile(downloadTask);
            }
        }, new ActionCameraCommandCallback1<YICameraSDKError>() {
            @Override
            public void onInvoke(YICameraSDKError yiCameraSDKError) {
                ambilFotoActivity.onCameraFailedDownloadFile(yiCameraSDKError);
            }
        });
    }
```

knkpxt, 2017-06-07:
Hi rahmatnazali!
I also have same problems. I successfully downloaded file through my app but got corrupted file. If you open file in binary mode you will see that data is scattered - there are empty spaces between data stream. Hope this issue will be handled fast

rahmatnazali, 2017-06-07:
Hi @knkpxt !
Somehow this is a nice things to know, meaning that I am not the only one and at least the error was not in our code then :))

Opening the files in binary mode was very neat observation, I never thought of that.

So do you think we just need to wait them to handle and can't do anything?

I'm just a newcomer so my skill is not that much, but these are what I already tried:
1. I tried to working on modifying the `ActionCamera.downloadFile`, but that means creating the whole API since those are all package-private so I don't think that was a valid solution :))
 2. I try to download [YI Action - YI Action Camera](https://play.google.com/store/apps/details?id=com.xiaomi.xy.sportscamera&hl=en) on Playstore and run it on my device. Surprisingly, the `download captured image` feature works like a charm and the downloaded image can also be opened. From this point, I am thinking to do a reverse engineering by decompile the YI Action - Action Camera APK, but in result I got a tons of PITA code :))

rahmatnazali, 2017-06-13:
To @knkpxt and other people having the same issue, I found the workaround! :)
This is not an elegant way, but at least this works.

Before moving further, you guys would need to find the address of the camera's IP, which is `mCameraIP` variable inside the API. In my case, I made a brand new custom api just to print/log the `mCameraIP`, and found that (ofcourse) the IP address is `192.168.42.1`, just like the IP we used to connect to the camera. I tested it for like 20 times and the camera IP address did not change, so I'm assuming it was a static address.

So after knowing where to download the files, I just use [Glide ](https://futurestud.io/tutorials/glide-getting-started) to download the image via android, using the code from [here](https://stackoverflow.com/questions/27394016/how-does-one-use-glide-to-download-an-image-into-a-bitmap), or like so:

```
Bitmap theBitmap = Glide.
        with(getBaseContext()).
        load("http://192.168.42.1/DCIM/100MEDIA/YDXJ00001.JPG").
        asBitmap().
        into(100, 100). // Width and height, fill it with (Target.SIZE_ORIGINAL, Target.SIZE_ORIGINAL) for the original size
        get();
```

you can then able to modify the image or even save it to external storage.

This solution makes me able to continue the project.

I hope this helps :)

Water-bamboo, 2017-06-14:
Will fix it soon.

rahmatnazali, 2017-06-14:
Hi @Water-bamboo , thanks for answering my issue :)

Super glad to hear you guys are on it. If you need guys another informations from me to provide regarding this issue, please let me know :)

closed 2017-06-14 by Water-bamboo

Water-bamboo, 2017-06-15:
Please pull new code from master branch.

rahmatnazali, 2017-06-15:
@Water-bamboo Thanks!
I already downloaded the newest JAR from master branch, if that is what you mean.