YITechnology/YIOpenAPI #50: Windows Prevents Multiple Yi 4K devices from being in U-Disk mode at once

somacoder, 2017-06-07:
I have a project that makes use of multiple Yi 4K cameras connected to a PC for syncing. However, Windows only allows a single camera to enter U-Disk mode at a time, regardless of being connected to two different USB bus ports. However, this is not the case on Mac OSX, which can connect multiple Yi 4K cameras in U-Disk mode at once. Is there a way to force Windows to allow multiple Yi 4K devices to enter U-Disk mode at once? Thank you!

Water-bamboo, 2017-06-07:
This may be can only be resolved by Windows.

somacoder, 2017-06-07:
Could it also be a driver issue? Are there any alternative Windows drivers for the Yi 4K U-disk mode?

somacoder, 2017-06-08:
The problem appears to be that all Yi 4k cameras show up as having the same "Device Instance Path". Since they are not unique, when the second one is plugged in, Windows fails to recognize it as a new device. More specifically, this may be related to the "serial number" suffix at the end of the instance path. The assignment of this should be uniquely fed from the device, but again, all Yi 4K devices appear to have the same one to Windows. Not sure if this could help anyone to garner a solution. I hope so. I've got nothing so far.

Here is the Instance Path for all Yi 4K cameras in Windows: 
USBSTOR\DISK&VEN_XIAOYI&PROD_SPORTCAMERA_2&REV_1000\0001&0

somacoder, 2017-06-09:
Solution: Change the USB Device Registry Entry for the device to Ignore the Hardware Serial Number and go with the USB Port instance to act as the suffix the the Instance ID.

closed 2017-06-09 by somacoder