YITechnology/YIOpenAPI #51: Firmware update issue

Erick-inestroza, 2017-06-17:
Hi
I tried to updated the firmware for my six cameras to on 360 video. The cameras didn't gave any message, insted it gave the video screen, but is black, and there's a red fix light on the top. When i tried to hit the button to record, they don't do anything. In the camera modes they allow me just photo and video. My question is: Is this normal? Or there was an error with the update? The batteries where fully charged.

Water-bamboo, 2017-06-26:
It sounds not normal. Which version of firmware do you get to upgrade?

Erick-inestroza, 2017-06-26:
I downloaded this version
https://github.com/YITechnology/YIOpenAPI/tree/master/vr/bin/1.0.179/en-us

Water-bamboo, 2017-07-04:
What‘s your camera version？ YI 4K Action Camera or YI 4K+ Action Camera?

zeratod, 2017-07-04:
4k and 4k+ 发自网易邮箱大师 在2017年07月04日 14:05，Water-bamboo 写道： What‘s your camera version？ YI 4K Action Camera or YI 4K+ Action Camera?

Erick-inestroza, 2017-07-04:
Mi camera version is the Yi 4K Action Camera

Samiboy007, 2017-07-23:
It worked for me. I've downloaded the firmwares. Rename firmware_first for the controller camera and firmware_others for the non-controller cameras. After the update, the battery level indicator won't longer be accurate.The screens are black when turned on but show the image when you start recording. I then link the sync cables that I bought from the Yi website. Turn on the bluetooth on the main camera and it works! The cameras start at the same time when I hit the record button but seems a little off when I press the stop button. The main camera seems to stop first, maybe one of frame after.