YITechnology/YIOpenAPI #54: 7 pin micro USB cable

guardfromua, 2017-07-21:
Hello!
In building VR camera we need 7 pin micro USB cable.
[image]
We can't wind it. Just in India and they don't ship it. May be we can buy it in Yi tech?
Because there is no link to the store. Just said, that you could make multicable by yourself.

Water-bamboo, 2017-07-31:
Sorry, We don't sell it.
Please find the more possible online store, for example aliexpress, taobao.com?.

guardfromua, 2017-07-31:
@Water-bamboo Thanks for your response.
Can u tell me the difference between 5 pin and 7 pin cable? Coz in another issue people connect those and multiending cable is not working.
PS: Will this cable work? Any suggestions? 
https://world.taobao.com/item/555445224213.htm?fromSite=main&spm=a230r.1.14.46.ebb2eb2a3nwOd&ns=1&abbucket=4#detail

qdkid, 2018-05-18:
would you please provide a taobao link for such cable? also what is the max # of cameras the SDK can support? Thank you.

qdkid, 2018-05-18:
found cable link here
https://store.yitechnology.com/products/yi-4k-action-camera-sync-cable-for-building-vr-camera