YITechnology/YIOpenAPI #55: Request for sourcecode for the build proces under the GPL

skinkie, 2017-07-23:
Firmware.bin contains Linux, Busybox, and Cherokee. It is obvious to state that this firmware is compiled from the Ambarella SDK. Under the GPL license I am requesting the used sourcecode and build process for the Xiaomi Yi camera's under a repository of YiTechnology. I am sure more people are interested in this.

`Linux version 3.10.71 (jenkins@ANTS-BJ-SV0003) (gcc version 4.9.2 20140904 (prerelease) (crosstool-NG linaro-1.13.1-4.9-2014.09 - Linaro GCC 4.9-2014.09) ) #1 PREEMPT Fri Sep 30 18:22:03 CST 2016`

So I guess it would be the most simple to state: Ambarella SoC SDK please :)

shali3, 2017-07-26:
+1 for source code.
You called this project YIOpenAPI, why isn't it open source?

akgordon, 2017-09-18:
I am finding some fundamental typos and bugs in this SDK that make this SDK unusable, but could easily be fixed if have source. Please make the source available.