YITechnology/YIOpenAPI #57: Xiaomi Yi 4K+ USB C Sync Cable for VR (genlock) build schema available?

archenroot, 2017-08-05:
Hello,

I just received 2 4k+ camera versions which are upgraded to USB C type of cable connectivity. Could you publish the schema for genlock sync, I can build my own cable, but need build info.

Thanks a lot.

chaohe1024, 2017-08-05:
4k+ doesn't support genlock

获取 Outlook for Android<https://aka.ms/ghei36>

发件人: archenroot
发送时间: 8月5日星期六 下午9:28
主题: [YITechnology/YIOpenAPI] Xiaomi Yi 4K+ USB C Sync Cable for VR (genlock) build schema available? (#57)
收件人: YITechnology/YIOpenAPI
抄送: Subscribed

Hello,

I just received 2 4k+ camera versions which are upgraded to USB C type of cable connectivity. Could you publish the schema for genlock sync, I can build my own cable, but need build info.

Thanks a lot.

archenroot, 2017-08-06:
@chaohe1024  Thanks for quick answer :-)

In my scenario for stereo vision I will try to fallback to streaming 2 independent streams life via WiFi router to PC, I have enough CPU threads to assign each stream to 1 thread and leave sync to another independent thread. On top I would like to do some kind of processing via GPU (I have 2 independent 1080 TI cards) via OpenCV.

No the question is which firmware is best for my purpose:
1 - standard one (1.2.8) available via https://www.yitechnology.com/firmware/index/class/action
2 - VR firmware available via https://github.com/YITechnology/YIOpenAPI/tree/master/vr/bin 

Maybe I don't need VR firmware as I will implement synchronization of images based on TIMESTAMP -> I will synchronize both cameras via NTP protocol.

I will definitelly apply little hacks to camera (get telnet, custom resolutions, etc.), but in general WHAT firmware is best.

Thanks for hints guys. Let explore full camera potential :-)

archenroot, 2017-08-08:
Closing this as nogo in the moment, will try to fix this on software level with NTP.

closed 2017-08-08 by archenroot

ruslanvasylev, 2019-04-06:
Hi archenroot,

Any news on this?