YITechnology/YIOpenAPI #64: getFileList always returns empty set

akgordon, 2017-09-13:
I am writing an Android application for the YI Action camera (not the 4K version) and have got a good start getting connected and viewing the live rtsp stream on phone, but when I try the getFileList API I always get the success case with empty set of results although I know there are files on the camera. Is there an example of how to use this API, or is other setup I needed? Thanks in advance for help.

akgordon, 2017-09-18:
Found problem, there is an error in the regular expression in the SDK. Please see issue #66

Asad1234567, 2018-06-20:
i can't edit the jar file please help me from  where i can edit this file and change the regular expression.