YITechnology/YIOpenAPI #67: Yi Lite sta.conf / synchronisation does not work

studio360de, 2017-09-28:
Hi there! I'd like to synchronize several Yi Lite cameras.

I did update the firmware and then loaded the sta.conf file from github. I changed the SSID, pw and country code, as said in the description. I loaded it onto a micro SD, put it into the cam, started the cam, set the frequency to 2,4 and switched the Wifi on. But still the cam shows the "old" cam-intern network and the standard password. Seems like the cam just does not recognize any changes. Ideas? 

I appreciate any help!!

Best wishes! Rodger

nfulinarajr, 2017-09-28:
@studio360de +1

dhammans, 2017-10-12:
+2 on this. There are several things going on.

1. The documentation isn't too clear if this only works on the 4k camera, or if it extends to the Lite as well.

2. Where are you placing the sta.conf? In the root of the SD card? Or are you supposed to put them in /sdk/bin/en-us?

nfulinara, 2017-10-17:
Thought I'd help shed some light since this repo doesn't seem to be active with support.

@studio360de 
I'm experiencing the same thing however, connecting to your personal wifi does work. While the wifi information stays the same, it will connect to your wifi assuming the SSID and Password are correct. You should see the wifi symbol blinking off and on. This is a good indication that it's working.

@dhammans 
1. Not sure as I'm only working with the Yi 4k Action Camera.
2. Insert your SD card into your machine. Open the drive. You are now at the root. Place the sta.conf file there. Make sure you have edited the file with a program like Notepadd++ or Sublime text to preserve Unix style line endings.

qdkid, 2018-05-25:
my camera shows the same "old" intern wifi hotspot information. However if you check your router, it shows the camera connect to loca router with IP ok.