YITechnology/YIOpenAPI #72: Video preview lag

Egorkansk, 2018-03-29:
hello, tell me please why video preview is laging, in your sample app, but oficial yi action app is not laging? How i can set preview resolution to fix this problem?

entalent, 2018-03-30:
I decompiled the official app, and I found that the official app used a totally different way to play the RTSP stream of camera preview. The sample app simply uses VideoView, while the official app uses a 3rd party library (I searched for that package name on google and library seem to be in https://github.com/fenghxfriend/Android-RTSP-Player). The preview lag of VideoView seems to depend on device model... my MI 5 has a huge lag of 11 second, but a MI MIX 2 has a lag of about 1s, both using the sample app. I decided to use other RTSP library to play preview.

Egorkansk, 2018-03-30:
Thank you, for answer

x4080, 2018-05-30:
Is it possible to zoom using video preview ? I'm thinking using cctv zoom lens, and video preview zoom for adjusting focus on the fly